GiorgosXou/NeuralNetworks #1: Why not work with 10 bit input data?

nazariiixa, 2019-08-03:
I have 10 bit input data like this 
const double inputs[110][8] = {
{540,131,48,3,0,0,0,0},
{624,167,63,15,0,0,0,0},
{736,224,96,31,0,0,0,0},...
but after learning output is the same for exemple
0.8215888
0.8215888
0.8215888
...
after i divide for 1024 i have data like this
const double inputs[110][8] = {
{0.52734375,0.1279296875,0.046875,0.0029296875,0,0,0,0},
{0.609375,0.1630859375,0.0615234375,0.0146484375,0,0,0,0},
{0.71875,0.21875,0.09375,0.0302734375,0,0,0,0},...
and it works perfectly...

How to do without dividing the input data???

GiorgosXou, 2019-08-07:
@nazariiixa  As a good start, i think it would be right to **replace double with float, unlless you have made any channge in the source code of library**

**If that wont work** try instead of having inputs as _"const float inputs[110][8] = {{**540**,131,48,3,0,0,0,0},..."_ ,  having inputs as _"const float inputs[110][8] = {{**540.0**,131.0,48.0,3.0,0.0,0.0,0.0,0.0},..."_

**If that also won't work** , it would be really helpfull if you could post some more info about the microcontroller or code [...]

nazariiixa, 2019-08-07:
The first two methods do not work, the output values ​​are the same. Yes, everything works with normalization, but this is a microcontroller, and there is always not enough processor time, and there’s also an extra division ... May be problem is in float datatype? As i know in arduino float=double="as large as 3.4028235E+38 and as low as -3.4028235E+38" only... not 1024...
microcontroller esp32, my code based on "Backpropagation_double_Xor" i change here only inputs, expectedOutput, and layers.

GiorgosXou, 2019-08-07:
@nazariiixa 

Thanks for the insight, (but i deleted my previous comment because i realised why the issue existed)

After Viewing my code i realised that it is because of no normalization of 10bit input data (most probably i will add this feature in future updates*) : [It has to do with the Backpropagation/Training-phase and the inputs](https://datascience.stackexchange.com/questions/13178/how-to-normalize-data-for-neural-network-and-decision-forest)

*Thanks you for pointing it out (: , i should have thought about it .. but its ok (:

(i will add an example of normalization here such you can work with)

nazariiixa, 2019-08-07:
Ok, thank you!

GiorgosXou, 2019-08-07:
it is "_return (Input_i - MinInput)/(**MaxInput**-MinInput);_ "
i had by mistake "_return (Input_i - **MaxInput**)/(MaxInput-MinInput);_"

_i've edited it though_