GiorgosXou/NeuralNetworks #5: if you do not define REDUCE_RAM_STATIC_REFERENCE

lewisrichardson2103, 2021-02-16:
Hi again,

If you do not define REDUCE_RAM_STATIC_REFERENCE then you have an issue with const declarations and the pointer to the neural network within the library (*me).

If I run the backprop double_XOR example (I am using a adafruit feather sense)  then I get the following error (no changes made to code) :
NeuralNetwork.h:732:18: error: invalid conversion from 'const NeuralNetwork*' to 'NeuralNetwork*' [-fpermissive]
  732 |             me = NN;
      |                  ^~
      |                  |
      |                  const NeuralNetwork*

if you include a definition for the reduced ram then this issue goes away

Thanks

Lewis

GiorgosXou, 2021-02-16:
Hello Lewis 👋,

### What version of arduino do you use and what version of gcc?
**If** you are using Arduino's IDE,  **try** going to **_"file>Preferences"_**, check **_"compilation"_** and  **_"Use accessibility features"_** boxes and **then** recompile/verify the sketch and post the Output here.

### About the "Const" & Issue
To be honest It's not necessary for the library to have const in this specific case of Layer's constructors but it also shouldn't have had an issue with _(or so i think)_ ... anyways **most probably i will fix it in the next days** [...]

### About B00010000 aka REDUCE_RAM_STATIC_REFERENCE 
Just for information purposes, it doesn't really slows CPU performance _(actually it might even make it faster by ~microseconds [i would say] and i think that it uses a bit less ROM in same specific cases too)_ **the reason i've set it by default not to be used is** because the only thing that really changes and makes a difference is related to **user/programmer's experience** _(compared to when it's not enabled/used)_, is just that the **user has to use it in an ambiguous way when** using multiple networks on the same sketch _(because *me is statically referenced and has to be redefined after the use of another network)_

**PS.** Thanks you ❤️ for bringing this issue to my awareness, i had no idea that it existed [...]
<sub>(Although it's a weird issue.. i was expecting a warning and no error but whatever lol)</sub>

GiorgosXou, 2021-02-25:
@lewisrichardson2103 I found the issue, **I didn't know that i was supposed to use:**

```c++
Layer(const unsigned int &NumberOfInputs, const unsigned int &NumberOfOutputs, NeuralNetwork * const NN ) // ✅
```
**instead of** _(what i was using before)_ :
```c++
Layer(const unsigned int &NumberOfInputs, const unsigned int &NumberOfOutputs,  const * NeuralNetwork NN ) // 🚫
```

**for the compiler to translate it into  a  "[const pointer to a NeuralNetwork-Object](https://stackoverflow.com/a/31331389)"**

<sub>Again thanks you ❤️ for bringing this issue to my awareness [...]</sub>

GiorgosXou, 2021-02-25:
( just saying, **found another issue**, i had forgot to add proper _"multiple activation function"_ support to Backprop _[just in case you used anything like it and you have other issues]_)

closed 2021-02-25 by GiorgosXou