GiorgosXou/NeuralNetworks #14: ESP32 Error "invalid conversion from 'const float*' to 'float*' [-fpermissive]"

kamon2022, 2023-02-14:
Hi, 
I have some problem if I use ESP32 
I found this error. "invalid conversion from 'const float*' to 'float*' [-fpermissive]" in file "FeedForward_double_Xor" 

Could you help me ? 
I can compile if i use arduino mega. I can not compile if I use esp32

GiorgosXou, 2023-02-15:
Just remove the `const` at those 2 lines:

https://github.com/GiorgosXou/NeuralNetworks/blob/31bb95e4cefcaefb28f6b031a634ded6c7274a7d/examples/FeedForward_double_Xor/FeedForward_double_Xor.ino#L22

https://github.com/GiorgosXou/NeuralNetworks/blob/31bb95e4cefcaefb28f6b031a634ded6c7274a7d/examples/FeedForward_double_Xor/FeedForward_double_Xor.ino#L25

[Reference](https://forum.arduino.cc/t/esp32-neural-network-library-const-float-to-float-fpermissive/1089929)

kamon2022, 2023-02-16:
Thank you very much for helping.

closed 2023-11-21 by GiorgosXou