GiorgosXou/NeuralNetworks #17: ESP32 error: 'constexpr' needed

MacrossGalang, 2023-10-25:
Hello Sir
I found this error in "any activation function per layer" example when compiling in esp32
however if I compile it using arduino uno it was just find

I have try to remove the "const" from the line, but still problem

C:\Users\macro\Documents\Arduino\libraries\NeuralNetwork\src/NeuralNetwork.h:444:9: error: 'activation_Function_ptrs' declared as an 'inline' field
         };
         ^
C:\Users\macro\Documents\Arduino\libraries\NeuralNetwork\src/NeuralNetwork.h:444:9: error: 'constexpr' needed for in-class initialization of static data member 'float (NeuralNetwork::Layer::* const NeuralNetwork::activation_Function_ptrs [2])(const float&)' of non-integral type [-fpermissive]

C:\Users\macro\Documents\Arduino\libraries\NeuralNetwork\src/NeuralNetwork.h:471:13: error: 'derivative_Function_ptrs' declared as an 'inline' field
             };
             ^
C:\Users\macro\Documents\Arduino\libraries\NeuralNetwork\src/NeuralNetwork.h:471:13: error: 'constexpr' needed for in-class initialization of static data member 'float (NeuralNetwork::Layer::* const NeuralNetwork::derivative_Function_ptrs [2])(const float&)' of non-integral type [-fpermissive]

GiorgosXou, 2023-11-01:
Hello @MacrossGalang, I tried both of the available board-packages and it compile just fine, I'm not sure how to reproduce your issue, could you add more informations about how you compile it and which exactly one esp32 you use?

[image]

MacrossGalang, 2023-11-03:
Hello Sir
the problem solve Sir
I just need to update the esp32 board-packages
from 2.0.8 to become 2.0.11 just like your screenshot above
thank you Sir

closed 2023-11-03 by GiorgosXou