GiorgosXou/NeuralNetworks #32: Resolve warnings: comparison between signed and unsigned integer

GiorgosXou, 2024-05-17:
How I messed-up signed and unsigned...

```
home/xou/Arduino/libraries/NeuralNetworks/src/NeuralNetwork.h: In member function 'void NeuralNetwork::pdestract()':
/home/xou/Arduino/libraries/NeuralNetworks/src/NeuralNetwork.h:1096:35: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
                 for (int i = 0; i < numberOflayers; i++)
                                 ~~^~~~~~~~~~~~~~~~
/home/xou/Arduino/libraries/NeuralNetworks/src/NeuralNetwork.h:1099:43: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
                         for (int j = 0; j < layers[i]._numberOfOutputs; j++) // because of this i wont make _numberOfOutputs/inputs private :/ or maybe.. i ll see... or i will change them to const* ... what? i've just read it again lol
                                         ~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/xou/Arduino/libraries/NeuralNetworks/src/NeuralNetwork.h:1132:39: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
                     for (int i = 0; i < numberOflayers; i++){
                                     ~~^~~~~~~~~~~~~~~~
/home/xou/Arduino/libraries/NeuralNetworks/src/NeuralNetwork.h: In constructor 'NeuralNetwork::NeuralNetwork(const unsigned int*, float*, float*, const unsigned int&, byte*)':
/home/xou/Arduino/libraries/NeuralNetworks/src/NeuralNetwork.h:1190:27: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         for (int i = 0; i < numberOflayers; i++)
                         ~~^~~~~~~~~~~~~~~~
/home/xou/Arduino/libraries/NeuralNetworks/src/NeuralNetwork.h: In constructor 'NeuralNetwork::NeuralNetwork(const unsigned int*, const unsigned int&, const float&, const float&, byte*)':
/home/xou/Arduino/libraries/NeuralNetworks/src/NeuralNetwork.h:1249:31: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
             for (int i = 0; i < numberOflayers; i++)
                             ~~^~~~~~~~~~~~~~~~
/home/xou/Arduino/libraries/NeuralNetworks/src/NeuralNetwork.h: In constructor 'NeuralNetwork::NeuralNetwork(const unsigned int*, const unsigned int&, byte*)':
/home/xou/Arduino/libraries/NeuralNetworks/src/NeuralNetwork.h:1280:31: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
             for (int i = 0; i < numberOflayers; i++)
                             ~~^~~~~~~~~~~~~~~~
/home/xou/Arduino/libraries/NeuralNetworks/src/NeuralNetwork.h: In member function 'float* NeuralNetwork::FeedForward_Individual(const float&)':
/home/xou/Arduino/libraries/NeuralNetworks/src/NeuralNetwork.h:1341:30: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         if (Individual_Input == layers[0]._numberOfInputs)
             ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/xou/Arduino/libraries/NeuralNetworks/src/NeuralNetwork.h:1377:31: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
             for (int i = 1; i < numberOflayers; i++)
                             ~~^~~~~~~~~~~~~~~~
/home/xou/Arduino/libraries/NeuralNetworks/src/NeuralNetwork.h: In member function 'float* NeuralNetwork::FeedForward(const float*)':
/home/xou/Arduino/libraries/NeuralNetworks/src/NeuralNetwork.h:1442:27: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         for (int i = 1; i < numberOflayers; i++)
                         ~~^~~~~~~~~~~~~~~~
/home/xou/Arduino/libraries/NeuralNetworks/src/NeuralNetwork.h: In member function 'void NeuralNetwork::print()':
/home/xou/Arduino/libraries/NeuralNetworks/src/NeuralNetwork.h:1723:27: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         for (int i = 0; i < numberOflayers; i++)
                         ~~^~~~~~~~~~~~~~~~
/home/xou/Arduino/libraries/NeuralNetworks/src/NeuralNetwork.h: In constructor 'NeuralNetwork::Layer::Layer(const unsigned int&, const unsigned int&, float*, float*, NeuralNetwork*)':
/home/xou/Arduino/libraries/NeuralNetworks/src/NeuralNetwork.h:1778:31: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
             for (int i = 0; i < _numberOfOutputs; i++)              // [matrix] (_numberOfOutputs * _numberOfInputs)
                             ~~^~~~~~~~~~~~~~~~~~
/home/xou/Arduino/libraries/NeuralNetworks/src/NeuralNetwork.h: In constructor 'NeuralNetwork::Layer::Layer(const unsigned int&, const unsigned int&, NeuralNetwork*)':
/home/xou/Arduino/libraries/NeuralNetworks/src/NeuralNetwork.h:1831:31: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
             for (int i = 0; i < _numberOfOutputs; i++)
                             ~~^~~~~~~~~~~~~~~~~~
/home/xou/Arduino/libraries/NeuralNetworks/src/NeuralNetwork.h:1841:35: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
                 for (int j = 0; j < _numberOfInputs; j++)
                                 ~~^~~~~~~~~~~~~~~~~
/home/xou/Arduino/libraries/NeuralNetworks/src/NeuralNetwork.h: In member function 'void NeuralNetwork::Layer::FdF_Individual_PROGMEM(const float&, const int&)':
/home/xou/Arduino/libraries/NeuralNetworks/src/NeuralNetwork.h:1886:23: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         for (i = 0; i < _numberOfOutputs; i++)
                     ~~^~~~~~~~~~~~~~~~~~
/home/xou/Arduino/libraries/NeuralNetworks/src/NeuralNetwork.h:1910:15: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         if (j == _numberOfInputs -1){
             ~~^~~~~~~~~~~~~~~~~~~~~
/home/xou/Arduino/libraries/NeuralNetworks/src/NeuralNetwork.h:1911:27: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
             for (i = 0; i < _numberOfOutputs; i++)
                         ~~^~~~~~~~~~~~~~~~~~
/home/xou/Arduino/libraries/NeuralNetworks/src/NeuralNetwork.h: In member function 'void NeuralNetwork::Layer::FeedForward_Individual(const float&, const int&)':
/home/xou/Arduino/libraries/NeuralNetworks/src/NeuralNetwork.h:1947:23: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         for (i = 0; i < _numberOfOutputs; i++)
                     ~~^~~~~~~~~~~~~~~~~~
/home/xou/Arduino/libraries/NeuralNetworks/src/NeuralNetwork.h:1971:15: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         if (j == _numberOfInputs -1){
             ~~^~~~~~~~~~~~~~~~~~~~~
/home/xou/Arduino/libraries/NeuralNetworks/src/NeuralNetwork.h:1972:27: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
             for (i = 0; i < _numberOfOutputs; i++){
[... 22 lines cut ...]
/home/xou/Arduino/libraries/NeuralNetworks/src/NeuralNetwork.h: In member function 'void NeuralNetwork::Layer::BackPropHidden(const NeuralNetwork::Layer*, const float*)':
/home/xou/Arduino/libraries/NeuralNetworks/src/NeuralNetwork.h:2445:35: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
                 for (int i = 0; i < _numberOfOutputs; i++)
                                 ~~^~~~~~~~~~~~~~~~~~
/home/xou/Arduino/libraries/NeuralNetworks/src/NeuralNetwork.h:2459:39: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
                     for (int j = 0; j < _numberOfInputs; j++)
                                     ~~^~~~~~~~~~~~~~~~~
/home/xou/Arduino/libraries/NeuralNetworks/src/NeuralNetwork.h: In member function 'void NeuralNetwork::Layer::print()':
/home/xou/Arduino/libraries/NeuralNetworks/src/NeuralNetwork.h:2492:27: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
         for (int i = 0; i < _numberOfOutputs; i++)
                         ~~^~~~~~~~~~~~~~~~~~
/home/xou/Arduino/libraries/NeuralNetworks/src/NeuralNetwork.h:2500:31: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
             for (int j = 0; j < _numberOfInputs; j++)
                             ~~^~~~~~~~~~~~~~~~~
/home/xou/Arduino/libraries/NeuralNetworks/src/NeuralNetwork.h: In member function 'void NeuralNetwork::Layer::print_PROGMEM()':
[... cut ...]
/home/xou/Arduino/libraries/NeuralNetworks/src/NeuralNetwork.h:2543:31: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
             for (int j = 0; j < _numberOfInputs; j++)
                             ~~^~~~~~~~~~~~~~~~~
/tmp/.arduinoIDE-unsaved2024417-15628-dgwrt5.w3c4/Run_In_Loop/Run_In_Loop.ino: In function 'void setup()':
/tmp/.arduinoIDE-unsaved2024417-15628-dgwrt5.w3c4/Run_In_Loop/Run_In_Loop.ino:28:23: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
     for (int j = 0; j < NumberOf(inputs); j++) // Epoch
```

GiorgosXou, 2024-05-29:
Ok, it might not be a good idea to resolve this issue considering there's for example this:
https://github.com/GiorgosXou/NeuralNetworks/blob/356d3ccd3a79b8d4a4b41a491644be0bbbbda97d/src/NeuralNetwork.h#L2341

noblip, 2024-07-25:
One option to replace
`for (int i = _numberOfOutputs -1; i >= 0; i--)`
replace with
```
for (unsigned int ii = _numberOfOutputs; ii > 0; ii--)
{
  auto i = ii - 1;
  //rest of the loop body is unchanged
```

GiorgosXou, 2024-07-25:
Thanks I'll probably fix this in couple of hours

GiorgosXou, 2024-07-25:
Actually I was wrong, I don't need to change that `i--` to unsigned at all. *(I mean.. I don't think anyone is going to train any neural network with 2,147,483,647 +1  layers, lol)*... Anyways, just fixing it right now

commit f685e9b4 (2024-07-25): FIXED: #32 #39 warnings (+improvement)
closed 2024-07-25 by GiorgosXou