GiorgosXou/NeuralNetworks #39: Lot of warnings about signed against unsigned comparison

noblip, 2024-07-25:
Example projects compiled for Teensy 4.0 target produce lots of warning about signed against unsigned comparison:

```
NeuralNetwork.h: In member function 'void NeuralNetwork::pdestract()':
NeuralNetwork.h:1122:27: warning: comparison of integer expressions of different signedness: 'int' and 'unsigned int' [-Wsign-compare]
 1122 |                     if (i == numberOflayers-1){ // -1 because we need final-outputs(below) to be managed by user.
      |                         ~~^~~~~~~~~~~~~~~~~~~
NeuralNetwork.h:1135:39: warning: comparison of integer expressions of different signedness: 'int' and 'unsigned int' [-Wsign-compare]
 1135 |                     for (int i = 0; i < numberOflayers -1; i++){ // -1 because we need final-outputs to be managed by user.
      |                                     ~~^~~~~~~~~~~~~~~~~~~
NeuralNetwork.h: In constructor 'NeuralNetwork::NeuralNetwork(const unsigned int*, float*, float*, const unsigned int&, byte*)':
NeuralNetwork.h:1194:27: warning: comparison of integer expressions of different signedness: 'int' and 'unsigned int' [-Wsign-compare]
 1194 |         for (int i = 0; i < numberOflayers; i++)
      |                         ~~^~~~~~~~~~~~~~~~
NeuralNetwork.h: In constructor 'NeuralNetwork::NeuralNetwork(const unsigned int*, const unsigned int&, const float&, const float&, byte*)':
NeuralNetwork.h:1249:35: warning: comparison of integer expressions of different signedness: 'int' and 'unsigned int' [-Wsign-compare]
 1249 |                 for (int i = 0; i < numberOflayers; i++)
      |                                 ~~^~~~~~~~~~~~~~~~
NeuralNetwork.h:1256:31: warning: comparison of integer expressions of different signedness: 'int' and 'unsigned int' [-Wsign-compare]
 1256 |             for (int i = 0; i < numberOflayers; i++)
      |                             ~~^~~~~~~~~~~~~~~~
NeuralNetwork.h: In constructor 'NeuralNetwork::NeuralNetwork(const unsigned int*, const unsigned int&, byte*)':
NeuralNetwork.h:1283:35: warning: comparison of integer expressions of different signedness: 'int' and 'unsigned int' [-Wsign-compare]
 1283 |                 for (int i = 0; i < numberOflayers; i++)
      |                                 ~~^~~~~~~~~~~~~~~~
NeuralNetwork.h:1290:31: warning: comparison of integer expressions of different signedness: 'int' and 'unsigned int' [-Wsign-compare]
 1290 |             for (int i = 0; i < numberOflayers; i++)
      |                             ~~^~~~~~~~~~~~~~~~
NeuralNetwork.h: In member function 'float* NeuralNetwork::FeedForward_Individual(const float&)':
NeuralNetwork.h:1357:30: warning: comparison of integer expressions of different signedness: 'int' and 'unsigned int' [-Wsign-compare]
 1357 |         if (Individual_Input == layers[0]._numberOfInputs)
      |             ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~
NeuralNetwork.h:1390:22: warning: comparison of integer expressions of different signedness: 'int' and 'unsigned int' [-Wsign-compare]
 1390 |             for (; i < numberOflayers; i++)
      |                    ~~^~~~~~~~~~~~~~~~
NeuralNetwork.h: In member function 'float* NeuralNetwork::FeedForward(const float*)':
NeuralNetwork.h:1457:18: warning: comparison of integer expressions of different signedness: 'int' and 'unsigned int' [-Wsign-compare]
 1457 |         for (; i < numberOflayers; i++)
      |                ~~^~~~~~~~~~~~~~~~
NeuralNetwork.h: In member function 'void NeuralNetwork::print()':
NeuralNetwork.h:1769:27: warning: comparison of integer expressions of different signedness: 'int' and 'unsigned int' [-Wsign-compare]
 1769 |         for (int i = 0; i < numberOflayers; i++)
      |                         ~~^~~~~~~~~~~~~~~~
NeuralNetwork.h: In constructor 'NeuralNetwork::Layer::Layer(const unsigned int&, const unsigned int&, NeuralNetwork*)':
NeuralNetwork.h:1985:31: warning: comparison of integer expressions of different signedness: 'int' and 'unsigned int' [-Wsign-compare]
 1985 |             for (int i = 0; i < _numberOfOutputs; i++)
      |                             ~~^~~~~~~~~~~~~~~~~~
NeuralNetwork.h:1995:35: warning: comparison of integer expressions of different signedness: 'int' and 'unsigned int' [-Wsign-compare]
 1995 |                 for (int j = 0; j < _numberOfInputs; j++)
      |                                 ~~^~~~~~~~~~~~~~~~~
NeuralNetwork.h: In member function 'void NeuralNetwork::Layer::FdF_Individual_PROGMEM(const float&, const int&)':
NeuralNetwork.h:2040:23: warning: comparison of integer expressions of different signedness: 'int' and 'unsigned int' [-Wsign-compare]
 2040 |         for (i = 0; i < _numberOfOutputs; i++)
      |                     ~~^~~~~~~~~~~~~~~~~~
NeuralNetwork.h:2064:15: warning: comparison of integer expressions of different signedness: 'const int' and 'unsigned int' [-Wsign-compare]
 2064 |         if (j == _numberOfInputs -1){
      |             ~~^~~~~~~~~~~~~~~~~~~~~
NeuralNetwork.h:2065:27: warning: comparison of integer expressions of different signedness: 'int' and 'unsigned int' [-Wsign-compare]
 2065 |             for (i = 0; i < _numberOfOutputs; i++)
      |                         ~~^~~~~~~~~~~~~~~~~~
NeuralNetwork.h: In member function 'void NeuralNetwork::Layer::FeedForward_Individual(const float&, const int&)':
NeuralNetwork.h:2101:23: warning: comparison of integer expressions of different signedness: 'int' and 'unsigned int' [-Wsign-compare]
 2101 |         for (i = 0; i < _numberOfOutputs; i++)
      |                     ~~^~~~~~~~~~~~~~~~~~
NeuralNetwork.h:2125:15: warning: comparison of integer expressions of different signedness: 'const int' and 'unsigned int' [-Wsign-compare]
 2125 |         if (j == _numberOfInputs -1){
      |             ~~^~~~~~~~~~~~~~~~~~~~~
NeuralNetwork.h:2126:27: warning: comparison of integer expressions of different signedness: 'int' and 'unsigned int' [-Wsign-compare]
 2126 |             for (i = 0; i < _numberOfOutputs; i++){
      |                         ~~^~~~~~~~~~~~~~~~~~
NeuralNetwork.h: In member function 'void NeuralNetwork::Layer::FdF_PROGMEM(const float*)':
NeuralNetwork.h:2287:27: warning: comparison of integer expressions of different signedness: 'int' and 'unsigned int' [-Wsign-compare]
 2287 |         for (int i = 0; i < _numberOfOutputs; i++)
      |                         ~~^~~~~~~~~~~~~~~~~~
NeuralNetwork.h:2297:31: warning: comparison of integer expressions of different signedness: 'int' and 'unsigned int' [-Wsign-compare]
 2297 |             for (int j = 0; j < _numberOfInputs; j++)
      |                             ~~^~~~~~~~~~~~~~~~~
NeuralNetwork.h: In member function 'void NeuralNetwork::Layer::FeedForward(const float*)':
NeuralNetwork.h:2332:27: warning: comparison of integer expressions of different signedness: 'int' and 'unsigned int' [-Wsign-compare]
 2332 |         for (int i = 0; i < _numberOfOutputs; i++)
      |                         ~~^~~~~~~~~~~~~~~~~~
NeuralNetwork.h:2356:35: warning: comparison of integer expressions of different signedness: 'int' and 'unsigned int' [-Wsign-compare]
 2356 |                 for (int j = 0; j < _numberOfInputs; j++)
      |                                 ~~^~~~~~~~~~~~~~~~~
NeuralNetwork.h: In member function 'void NeuralNetwork::Layer::print()':
NeuralNetwork.h:2649:27: warning: comparison of integer expressions of different signedness: 'int' and 'unsigned int' [-Wsign-compare]
 2649 |         for (int i = 0; i < _numberOfOutputs; i++)
      |                         ~~^~~~~~~~~~~~~~~~~~
NeuralNetwork.h:2657:31: warning: comparison of integer expressions of different signedness: 'int' and 'unsigned int' [-Wsign-compare]
 2657 |             for (int j = 0; j < _numberOfInputs; j++)
      |                             ~~^~~~~~~~~~~~~~~~~
NeuralNetwork.h: In member function 'void NeuralNetwork::Layer::print_PROGMEM()':
NeuralNetwork.h:2696:27: warning: comparison of integer expressions of different signedness: 'int' and 'unsigned int' [-Wsign-compare]
 2696 |         for (int i = 0; i < _numberOfOutputs; i++)
      |                         ~~^~~~~~~~~~~~~~~~~~
NeuralNetwork.h:2704:31: warning: comparison of integer expressions of different signedness: 'int' and 'unsigned int' [-Wsign-compare]
 2704 |             for (int j = 0; j < _numberOfInputs; j++)
      |                             ~~^~~~~~~~~~~~~~~~~
```

GiorgosXou, 2024-07-25:
7am here,  just woke  up 😪, I'll do my morning routine and then fix this. Gm btw

GiorgosXou, 2024-07-25:
Doooneeeeeee

closed 2024-07-25 by GiorgosXou