GiorgosXou/NeuralNetworks #47: Training backprop throws NaN into the model

Shirkit, 2025-03-13:
Hello @GiorgosXou ,

I've created a minimal branch (https://github.com/Shirkit/atlantico-device/tree/erroring) that is displaying a behavior that I cannot explain. 

When a model (randomly assigned weights and biases) parses though the second line in the CSVs (the one that starts on -1 in the X file) the model fails to process that line and NaN's in the weights and biases. This does not happen when I have the precomputed model ran through the same exact process. 

I cannot explain why this is happening, I tried my best and I can't find the root issue there. Do you mind taking a look? I'm using a standard ESP32 on it. If you have PlatformIO this project is just a build and run.

Shirkit, 2025-03-13:
If you don't have an ESP32 I can try to come up with something else for you to run. But with an ESP32 and PlatformIO it's just an Upload and Monitor and it just runs. Anyways, there's the output of the program.

<summary>
--- Terminal on /dev/ttyUSB0 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
Training model from original dataset...
 ========= Input model =========

----------------------
20x16| F(x):0
   B:0.7659330
1  W: 0.1265510  W: 0.9504270  W: 0.5161900  W: 0.5426240  W: 0.0406350  W: 0.6049200  W: 0.4279440  W: 0.7193100  W: 0.2698620  W: 0.9984840  W: 0.9373630  W: 0.6433010  W: 0.9147100  W: 0.2672360  W: 0.3518760  W: 0.7504310  W: 0.5799290  W: 0.5603970  W: 0.5006750  W: 0.9704910 
   B:0.3777430
2  W: 0.2241900  W: 0.0593440  W: 0.4581340  W: 0.2684250  W: 0.8416290  W: 0.8440300  W: 0.2527270  W: 0.4291260  W: 0.1417430  W: 0.6873340  W: 0.9341040  W: 0.6747600  W: 0.6297490  W: 0.9656200  W: 0.2302560  W: 0.6709320  W: 0.0355720  W: 0.5316130  W: 0.4024900  W: 0.9545090 
   B:0.2162620
3  W: 0.1271190  W: 0.3374050  W: 0.8756950  W: 0.6540490  W: 0.4303270  W: 0.6567190  W: 0.0334970  W: 0.5668240  W: 0.0925960  W: 0.2186490  W: 0.2873560  W: 0.6541840  W: 0.6230930  W: 0.2222450  W: 0.5810070  W: 0.0993060  W: 0.4265090  W: 0.4427540  W: 0.5803520  W: 0.4256650 
   B:0.3915290
4  W: 0.2167830  W: 0.6387380  W: 0.1708010  W: 0.9796900  W: 0.1613300  W: 0.8873370  W: 0.2281950  W: 0.6366560  W: 0.0959630  W: 0.0820110  W: 0.2994270  W: 0.8400420  W: 0.4871060  W: 0.4009680  W: 0.2062120  W: 0.7590010  W: 0.8345100  W: 0.8824800  W: 0.2846580  W: 0.8865710 
   B:0.6697000
5  W: 0.9833310  W: 0.0008140  W: 0.7848470  W: 0.1705640  W: 0.0131970  W: 0.1036250  W: 0.7484380  W: 0.4609310  W: 0.9980180  W: 0.9984870  W: 0.5701510  W: 0.1691870  W: 0.4099130  W: 0.3241790  W: 0.0269950  W: 0.6717500  W: 0.4117500  W: 0.3649130  W: 0.2955620  W: 0.9311340 
   B:0.4755080
6  W: 0.9622730  W: 0.2435470  W: 0.4185210  W: 0.3048300  W: 0.1591400  W: 0.9255570  W: 0.7776780  W: 0.8547260  W: 0.9385030  W: 0.5505970  W: 0.3204080  W: 0.8848930  W: 0.2139880  W: 0.3392380  W: 0.1170850  W: 0.4400930  W: 0.6771880  W: 0.7427200  W: 0.4832110  W: 0.9587460 
   B:0.3497520
7  W: 0.3323870  W: 0.8157100  W: 0.0792090  W: 0.5958870  W: 0.8586680  W: 0.5831030  W: 0.1234730  W: 0.0719000  W: 0.2076740  W: 0.3751050  W: 0.7961830  W: 0.1409520  W: 0.9843700  W: 0.1517870  W: 0.1593020  W: 0.7274100  W: 0.0219050  W: 0.5121100  W: 0.3054000  W: 0.1579960 
   B:0.8872560
8  W: 0.9501420  W: 0.8985850  W: 0.1918600  W: 0.7970470  W: 0.1439240  W: 0.6857310  W: 0.3211580  W: 0.6753860  W: 0.5352190  W: 0.3164000  W: 0.0434150  W: 0.0040550  W: 0.4816820  W: 0.9698740  W: 0.6430610  W: 0.1150060  W: 0.2836020  W: 0.1892680  W: 0.6893650  W: 0.0504700 
   B:0.4172560
9  W: 0.1365180  W: 0.0267230  W: 0.3760890  W: 0.3491060  W: 0.6943190  W: 0.0031300  W: 0.2476550  W: 0.8187320  W: 0.2939930  W: 0.5599740  W: 0.1275950  W: 0.6888100  W: 0.8236730  W: 0.8450550  W: 0.5087300  W: 0.2560950  W: 0.5582450  W: 0.0307050  W: 0.1476940  W: 0.2229480 
   B:0.1581190
10  W: 0.5598730  W: 0.9283530  W: 0.9344740  W: 0.5207600  W: 0.3807410  W: 0.5267560  W: 0.1556440  W: 0.7231430  W: 0.5316400  W: 0.1837370  W: 0.2846790  W: 0.1761350  W: 0.9011490  W: 0.8215290  W: 0.4272100  W: 0.3866070  W: 0.2244170  W: 0.2552550  W: 0.9537870  W: 0.9093730 
   B:0.6997110
11  W: 0.7076260  W: 0.7074450  W: 0.8841120  W: 0.0791720  W: 0.8135950  W: 0.8271530  W: 0.0550530  W: 0.5051830  W: 0.1617170  W: 0.9146550  W: 0.8460170  W: 0.9091760  W: 0.3330770  W: 0.2387760  W: 0.5153990  W: 0.3506540  W: 0.7927280  W: 0.6230020  W: 0.4821610  W: 0.9054550 
   B:0.2683510
12  W: 0.7088700  W: 0.9683160  W: 0.7356940  W: 0.0193810  W: 0.7579300  W: 0.8074800  W: 0.1266040  W: 0.7533860  W: 0.8613590  W: 0.4762070  W: 0.0419540  W: 0.2560790  W: 0.3737020  W: 0.4213650  W: 0.2744530  W: 0.7037320  W: 0.4398360  W: 0.1684320  W: 0.8945630  W: 0.7230170 
   B:0.7728430
13  W: 0.4351070  W: 0.9835830  W: 0.8149100  W: 0.1084670  W: 0.1481160  W: 0.6715460  W: 0.5517190  W: 0.7255280  W: 0.3480710  W: 0.0801070  W: 0.1782560  W: 0.4585200  W: 0.9223530  W: 0.9765280  W: 0.8881540  W: 0.0564300  W: 0.8734770  W: 0.6950480  W: 0.8465200  W: 0.9312040 
   B:0.0787050
14  W: 0.4834140  W: 0.5808100  W: 0.8357190  W: 0.9479250  W: 0.2582980  W: 0.3321310  W: 0.0117010  W: 0.7267090  W: 0.1307510  W: 0.3050220  W: 0.5128170  W: 0.5486520  W: 0.4739240  W: 0.9346950  W: 0.8604710  W: 0.1611530  W: 0.0046250  W: 0.2309780  W: 0.3556160  W: 0.5011420 
   B:0.7541080
15  W: 0.3465970  W: 0.3570880  W: 0.3417320  W: 0.8319170  W: 0.6444040  W: 0.6927430  W: 0.6803950  W: 0.5427490  W: 0.6544070  W: 0.5559360  W: 0.3743020  W: 0.8837670  W: 0.1660450  W: 0.7515630  W: 0.8303400  W: 0.0333980  W: 0.4448960  W: 0.6130510  W: 0.2878070  W: 0.7517670 
   B:0.5463930
16  W: 0.4049970  W: 0.1789780  W: 0.0806500  W: 0.3278090  W: 0.5623300  W: 0.0923600  W: 0.5307160  W: 0.0651610  W: 0.2522930  W: 0.7632090  W: 0.3237530  W: 0.8460980  W: 0.6651300  W: 0.9796340  W: 0.5345430  W: 0.1718890  W: 0.5740880  W: 0.1790310  W: 0.3006920  W: 0.9204010 
----------------------
16x8| F(x):0
   B:0.1183300
[... cut ...]
4  W:-0.6220031  W:-0.2261024  W: 1.0161490  W: 0.6378892  W:-1.1841401  W: 0.0319451  W: 0.5683460  W:-0.2061252 
   B:0.2811170
5  W:-0.7713876  W: 0.4469776  W: 0.1426064  W:-0.2630965  W:-0.5827761  W: 0.1227368  W:-0.7259946  W:-0.6209896 
   B:0.0081802
6  W: 0.1333522  W: 0.2294986  W: 0.2143248  W: 0.0800895  W: 0.7281337  W:-0.8326154  W:-0.6897705  W: 0.1004636 
----------------------
2

Choose an option to coninue:
1. Print Model
2. Train Model
3. Save Model
4. Load Model
5. Print New Model
6. Train New Model
7. Save New Model
8. Load New Model
9. Send Current Model to Network
</summary>

GiorgosXou, 2025-03-14:
👋 Hey @Shirkit, good morning *(from here)*

> When a model *(**randomly** assigned weights and biases)*...

**First of all** lets begin from something important, I figured out reacently too. When you deal with MCUs that utilize by default a [TRNG](https://en.wikipedia.org/wiki/Hardware_random_number_generator) *(True Random Number Generator)*, such as ESP32-MCUs **it's best for debuging-purposes\reproducibility to fall-back to the PRNG-algorithm** *(Psudo\Fake Random Number Generator)* **by simply using a deterministic `randomSeed(1)`** ([see source](https://github.com/esp8266/Arduino/pull/2142))

> Do you mind taking a look?

😄 Of course, Happy to help! **Before taking a deeper look, I have a few small suggestions.** wherever you use:
```cpp
#ifdef USE_64_BIT_DOUBLE
    double xy;
#else
    float xy;
#endif
```
**You may want to**, simply change those lines with just:
```cpp
DFLOAT xy; // meaning double OR float based on USE_64_BIT_DOUBLE
```

>  the model fails to process that line and NaN's in the weights and biases. This does not happen when I have the precomputed model ran through the same exact process. I cannot explain why this is happening...

🕵️‍♂️ **From a brief look in your code, everything seems great... Except!,** as said [here](https://github.com/Shirkit/atlantico-device/pull/1#issuecomment-2722292270) **`ReLU` is strongly susceptible to gradient-explosions**, and for that reason, you may **try**:

-  lowering `NN.LearningRateOfWeights` & `NN.LearningRateOfBiases` [(see also)](https://www.reddit.com/r/MachineLearning/comments/89vg84/comment/dwtwldo/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button)
-  [Decrease your initialization variances](https://www.reddit.com/r/MachineLearning/comments/89vg84/comment/dwu8b6u/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button)
- using other activation function
- etc. <sup><sub>(might think of something else later)</sub></sup>

>  I tried my best and I can't find the root issue there.

❤️ If you still don't find a way, let me know. I'm always happy to help. As of now, I'm developing the next version of this library which will support [`SimpleRNN`s](https://www.tensorflow.org/api_docs/python/tf/keras/layers/SimpleRNN) too! So, stay tuned for great things.

<sub>*(**tl;dr:** read just the bold lines 😅)*</sub>

GiorgosXou, 2025-03-14:
also fixed another potential-leak https://github.com/GiorgosXou/NeuralNetworks/commit/0c295c8ae3b48fbd31d77f3cce7d55d1c4b7eb77

Shirkit, 2025-03-15:
Aaahh, somehow I missed that Ovf and NaN comment you left me! Now everything is clear, I thought I'd face gradient explosion but it would be as Ovf (I actually faced this and imagined that was it a while ago), but not as NaN.

I'll be continuing the work from here! Thank you!

closed 2025-03-15 by Shirkit

Shirkit, 2025-03-15:
I have an unrelated question: in other libraries, we can set a `batch size` and pass `input layer * batch size` to the feedforward and backpropagation functions, and it goes in batches instead of one at a time.

I saw you referencing batch size (https://github.com/GiorgosXou/NeuralNetworks?tab=readme-ov-file#dfloat-loss-functions), but when I took a look at the code I didn't see it handling a batch size > 1. Does it support?

Btw,  if it's not supported, I'm not asking for support to be added, just asking since I scrapped that implementation.

GiorgosXou, 2025-03-15:
By the term batch-size, I mean inputs-per-epoch. eg. in [this `Backpropagation_Single_Xor` example](https://github.com/GiorgosXou/NeuralNetworks/blob/master/examples/Basic/Backpropagation_Single_Xor/Backpropagation_Single_Xor.ino) the batch-size is `NumberOf(inputs)` which is `4`, since it `FeedForward`s 4*2 inputs

Shirkit, 2025-03-20:
What's most interesting is that in that example, if I don't pass an `*default_Weights` and `*default_Bias` to the constructor function, the model does not explode. But if I set to anything, even full of zeroes, random values, a constant, I tried lots of different ways, none of them works, only if I don't set those initial vectors.

Shirkit, 2025-03-20:
No need, as always I'm to blame. The array of `default_Weights` and `default_Bias` are pointers and isAllocdWithNew is being set to false, so it's not copying the values.