GiorgosXou/NeuralNetworks #49: Improvement for activation functions

noblip, 2025-04-23:
To specify activation functions one has to use absolute indexes that change when activation function type used in user code change. So one has to remember to update numbers and order inside _Actv_Functions_ array. This can be improved to make it order independent and easier to maintain by adding the following enum into NeuralNetwork.h file. Additionally if one changes activation function defines  (like #define SELU) but forgets to update _Actv_Functions_ array compiler will stop with an error forcing to fix this. As an example declaration of _Actv_Functions_ from Any_Activation_Function_Per_Layer example will look like:

`byte Actv_Functions[] = {   Tanh_Idx, Tanh_Idx, Sigmoid_Idx}; // 1 = Tanh and 0 = Sigmoid (just as a proof of consept)  `

Code to add into NeuralNetwork.h:

```
enum 
{
#if defined(ALL_ACTIVATION_FUNCTIONS) or defined(Sigmoid)
Sigmoid_Idx,
#endif
#if defined(ALL_ACTIVATION_FUNCTIONS) or defined(Tanh)
Tanh_Idx,
#endif
#if defined(ALL_ACTIVATION_FUNCTIONS) or defined(ReLU)
ReLU_Idx,
#endif
#if defined(ALL_ACTIVATION_FUNCTIONS) or defined(LeakyReLU)
LeakyReLU_Idx,
#endif
#if defined(ALL_ACTIVATION_FUNCTIONS) or defined(ELU)
ELU_Idx,
#endif
#if defined(ALL_ACTIVATION_FUNCTIONS) or defined(SELU)
SELU_Idx,
#endif
#if defined(ALL_ACTIVATION_FUNCTIONS) or defined(Softmax)
SoftmaxSum_Idx,
#endif
#if defined(ALL_ACTIVATION_FUNCTIONS) or defined(Identity)
Identity_Idx,
#endif
#if defined(ALL_ACTIVATION_FUNCTIONS) or defined(BinaryStep)
BinaryStep_Idx,
#endif
#if defined(ALL_ACTIVATION_FUNCTIONS) or defined(Softplus)
Softplus_Idx,
#endif
#if defined(ALL_ACTIVATION_FUNCTIONS) or defined(SiLU)
SiLU_Idx,
#endif
#if defined(ALL_ACTIVATION_FUNCTIONS) or defined(GELU)
GELU_Idx,
#endif
#if defined(ALL_ACTIVATION_FUNCTIONS) or defined(Mish)
Mish_Idx,
#endif
#if defined(ALL_ACTIVATION_FUNCTIONS) or defined(Gaussian)
Gaussian_Idx,
#endif

#if defined(CUSTOM_AF1)
CUSTOM_AF1_Idx,
#endif
#if defined(CUSTOM_AF2)
CUSTOM_AF2_Idx,
#endif
#if defined(CUSTOM_AF3)
CUSTOM_AF3_Idx,
#endif
#if defined(CUSTOM_AF4)
CUSTOM_AF4_Idx,
#endif
#if defined(CUSTOM_AF5)
CUSTOM_AF5_Idx,
#endif
};
```

GiorgosXou, 2025-04-23:
Indeed sometimes they may feel like [Magic numbers](https://en.wikipedia.org/wiki/Magic_number_(programming). I'll implement them as `enum`s or macros and co-author you in the commit once I find time. Thanks for the idea.

GiorgosXou, 2025-08-19:
[image]