GiorgosXou/NeuralNetworks #50: Problems using large NN

noblip, 2025-04-23:
After having problems training my NN of about 200 inputs I wrote code that tries to train a written digit recognition NN based on well known MNIST data that uses NN of 784 inputs. However I could not make it work with code returning same MSE value after every epoch. Will you be interested in a minimal test code that demonstrates this issue?

GiorgosXou, 2025-04-23:
> Will you be interested in a minimal test code that demonstrates this issue?

Sure, I'd love to if that's possible. I'm really interested to see where the issue might rise from.

GiorgosXou, 2025-12-07:
**The issue might be due to heap fragmentation.** I may re-implement a less memory-efficient version that relies more on the stack, too.

**I haven’t tested training large neural networks on MCUs yet,** but there is a possible explanation I forgot to mention earlier that might apply to your case. The current backpropagation algorithm is designed with:

>  [...strict RAM optimization that allows the library to use one array that stores only the values after the activation instead of two arrays storing values before and after the activation, the use of some derivative functions in backpropagation](https://github.com/GiorgosXou/NeuralNetworks?tab=readme-ov-file#dfloat-activation-functions)